jsaintrocc/myblvmgncube #3: Missing 2 M3x10 mm in step 8?

marutter, 2021-05-22:
I believe in both back corners, there is a place to attach the stepper motor to the corner block with a M3x10mm.  Best to install them before the belts, I think.

Great set of pages.  Referencing them all the time while building my own MGN Cube.

jsaintrocc, 2021-05-23:
Hi @marutter thanks for the feedback!

I can't figure out where the 2 M3x10mm would attach to the steppers in [Step 8 Partially Build Back Corners](https://github.com/jsaintrocc/myblvmgncube/blob/master/manual/guide/08-back-corners-a.md). In fig 8.17 (Substep 11) we temporarily attach them using 2 M3x35mm. Which figure would be closest to where they should go?

~James

marutter, 2021-05-23:
In figure 8.6, up an to the right of the second arrow there is a 3mm hole countersunk for the socket head of a M3.  I don't think you refer to that location in the rest of the assembly instructions.

jsaintrocc, 2021-05-23:
Yup. Sure enough. I just pulled apart my left back corner and I completely missed it. Looks like an M3x8mm will be more in line with the other 3 screws. They only go about 3mm into the stepper. I'll add that to the instructions. Good catch!

jsaintrocc, 2021-05-23:
OK. Added the missing screws in Commit: f75bb6edfb0c274f0bd16ca5e63dc5460b07fc54

closed 2021-05-23 by jsaintrocc