jsaintrocc/myblvmgncube #10: Missing Y-endstop

marutter, 2021-08-14:
I just finished assembling my Cube through your step 14.  Your instructions have been amazing, thank you.

One thing I think you are missing is the Y endstop (part Y_Endstop_-_Front_Left_-_Rear_left).  Not sure if you are going to add that in a later step, or if should be added to step 12 and then adjusted later.

jsaintrocc, 2021-08-16:
Thanks @marutter. Good catch! I was debating where to put that but I like your suggestion of putting it in step 12 since it attaches to one of those cross beams. I'm add it once I'm done with my current step which is the cable management and extruder mounting for the hotend.

Regarding the 3d printed piece. Did you need to modify it at all too attach the switch? The holes look really small compared to the limit switch.

Thanks, James

jsaintrocc, 2021-08-16:
Did a quick test fit. Looks like M2 12mm screws w/washers will do but the 3d parts holes don't line up with switch.

jsaintrocc, 2021-08-17:
See Fysetc's [video](https://www.youtube.com/watch?v=KnPkUY0QOnw) at 16:18 to see how they did it.

jsaintrocc, 2021-09-13:
Ready to go just blocked by heated bed assembly.

PR #20 (2021-10-06): 10 missing y endstop
Closes #10